STMicroelectronics/STM32CubeL0 #3: Wrong frequencies on LL_RCC_GetSystemClocksFreq() if HSI16DIV flag is set

BatuuKurt, 2019-07-08:
**Caution**
The Issues are strictly limited for the reporting of problem encountered with the software provided in this project.
For any other problem related to the STM32 product, the performance, the hardware characteristics and boards, the tools the environment in general, please post a topic in the [ST Community/STM32 MCUs forum](https://community.st.com/s/group/0F90X000000AXsASAW/stm32-mcus) 

**Describe the set-up**
 * The board (either ST RPN reference or your custom board)
 * IDE or at least the compiler and its version

**Describe the bug**
A clear and concise description of what the bug is.

**How To Reproduce**
1. Indicate the global behavior of your application project

2. The modules that you suspect to be the cause of the problem (Driver, BSP, MW ...)

3. The use case that generates the problem

4. How we can reproduce the problem

**Additional context**
If you have a first analysis or patch correction, thank you to share your proposal.

**Screenshots**
If applicable, add screenshots to help explain your problem.

ALABSTM, 2019-07-10:
Hi Kurt,

Thank you for pointing out this issue. Actually, we found it out too and fixed it in release v1.**11**.0. Currently, release v1.**11.2** is available on GitHub. 

The code snippet below shows how the fix has been implemented (almost the same way as you did):
https://github.com/STMicroelectronics/STM32CubeL0/blob/65f3157a89766c2f89442115c49d29df73485eb9/Drivers/STM32L0xx_HAL_Driver/Src/stm32l0xx_ll_rcc.c#L590-L599

Could it be possible that you are using a firmware release prior to v1.11.0? Please let us know if it is the case and thank you once more for your contribution.

With regards,

ALABSTM, 2019-07-10:
ST Internal Reference: 32221

BatuuKurt, 2019-07-10:
Ali,
thanks for the hint. An update to v1.11.2 fixed the issue.

closed 2019-07-10 by BatuuKurt

ALABSTM, 2019-07-10:
Hi Kurt,

Really glad to hear that.