STMicroelectronics/STM32CubeL0 #4: 32L0538DISCOVERY Missing project files

K2AAE, 2020-04-09:
In STM32CubeL0/Projects/32L0538DISCOVERY/Demonstrations/ there are only project files for EWARM and MDK-ARM, no STM32CubeIDE (ST) or SW4STM32 (ac6) or TrueSTUDIO (Atollic/ST)

ALABSTM, 2020-04-14:
Hi @K2AAE,

Thank you for having reported this point. It has been forwarded to our development team. Hopefully SW4STM32 project files will be provided in the frame of a future release.

Unfortunately, we cannot share a date for the moment. We count on your patience and comprehension. Take care and stay safe.

With regards,

ALABSTM, 2020-04-14:
ST Internal Reference: 84588

ALABSTM, 2021-03-08:
Hi @K2AAE,

I hope you are fine. Unfortunately, we could not address your request and we do apologize for the inconvenience. Indeed, this is a known limitation reported in the [release note](https://htmlpreview.github.io/?https://github.com/STMicroelectronics/STM32CubeL0/blob/master/Release_Notes.html) (as shown in the screenshot below).

According to our development teams, there is an issue particularly with the SW4STM32 project of the demo that could not be solved. Hence, choice has been made not to provide it.

We do apologize once again for the inconvenience and we thank you in advance for your comprehension.

With regards,
Ali

[image]

closed 2021-03-08 by ALABSTM