STMicroelectronics/STM32CubeL0 #6: Compilation warning: redundant redeclaration of 'PLLMulTable' [-Wredundant-decls]

bbooher, 2020-06-23:
**Describe the set-up**
Using arm-none-eabi-gcc and custom makefile project using an STM32L0 within a packaged muRata LoRa module on a custom board.

**Additional context**
I believe the check and declaration in "stm32l0xx_hal_rcc_ex.c" lines 44-46 to be redundant, as PLLMulTable is declared in system_stm32l0xx.h line 63. It doesn't appear to cause an issue (I haven't yet checked for duplicate objects in the build output) but produces a compiler warning, as it should.

RKOUSTM, 2020-07-02:
Hi @bbooher,

Thank you for having reported this point. It will be forwarded to our development teams for analysis. We will be back to you as soon as we get feedback.

Thank you for your patience and thank you again for your contribution.

With regards,

RKOUSTM, 2020-07-22:
Hi @bbooher,

The **PLLMulTable** is declared external in ``stm32l0xx_hal_rcc_ex.c``, so there are no duplication.
 
https://github.com/STMicroelectronics/STM32CubeL0/blob/904c92833dc000c5819504d230c29e882d18f691/Drivers/STM32L0xx_HAL_Driver/Src/stm32l0xx_hal_rcc_ex.c#L44-L46

Please allow me to close this issue now. Thank you again for your report.

With regards,

closed 2020-07-22 by RKOUSTM