STMicroelectronics/STM32CubeL0 #7: __HAL_TSC_GET_GROUP_STATUS returns status for incorrect TSC group

austinphillips2, 2020-09-07:
The __HAL_TSC_GET_GROUP_STATUS macro is used to return the conversion status of an individual TSC group.

```#define __HAL_TSC_GET_GROUP_STATUS(__HANDLE__, __GX_INDEX__) ...```

The `__GX_INDEX__` field is incorrectly masked in the `#define` definition resulting in the status being returned for TSC group 0 rather than the group requested by `__GX_INDEX__`.

This issue has previously been reported and acknowledged at https://community.st.com/s/question/0D50X0000ASrqJjSQJ/bug-in-stm32l0-tsc-library but a fix has not been made available in this repository.

RKOUSTM, 2020-09-08:
Hi @austinphillips2,

Thank you for your suggestion and contribution. A fix  for this code below :

https://github.com/STMicroelectronics/STM32CubeL0/blob/904c92833dc000c5819504d230c29e882d18f691/Drivers/STM32L0xx_HAL_Driver/Inc/stm32l0xx_hal_tsc.h#L574-L575

will be published as soon in in future release (```STM32L0Cube V1.12.0```) looks like the following: 

```diff
+ #define __HAL_TSC_GET_GROUP_STATUS(__HANDLE__, __GX_INDEX__) \ 
+ ((((__HANDLE__)->Instance->IOGCSR & (uint32_t)(1UL << (((__GX_INDEX__) & 0xFUL) + 16UL))) == (uint32_t)(1UL << (((__GX_INDEX__) & 0xFUL) + 16UL))) ? TSC_GROUP_COMPLETED : TSC_GROUP_ONGOING)
```
Thank you again for your contribution.

With regards,

RKOUSTM, 2020-09-08:
ST Internal Reference: 62853

RKOUSTM, 2021-03-03:
Hi @austinphillips2,

I hope you are fine. The issue you reported has been fixed in the frame of version v1.12.0 of the STM32CubeL0 published on GitHub.

Thank you again for having reported.

With regards,

closed 2021-03-03 by RKOUSTM