STMicroelectronics/STM32CubeL0 #11: __HAL_RTC_TAMPER_GET_IT() fail

CanastraRF, 2020-10-02:
When I use 
__HAL_RTC_TAMPER_GET_IT()
with ARMCC5 I get a syntax error.

Please fix this.

ALABSTM, 2020-10-13:
Hi @CanastraRF,

Thank you for this report. Would you mind providing the error message you get in order to allow us to investigate this issue?

With regards,

CanastraRF, 2020-10-13:
Dear Support
I've got the following code snipped  for STM32L053:

  {
    RTC_HandleTypeDef hRtc;

    int i = __HAL_RTC_TAMPER_GET_IT(&hRtc, RTC_IT_TAMP2);
  }
The Compiler output is:

../Src/main.c(95): error:  #53: expected a ":"
      int i = __HAL_RTC_TAMPER_GET_IT(&hRtc, RTC_IT_TAMP2);
../Src/main.c: 0 warnings, 1 error
"../Src/main.c" - 1 Error(s), 0 Warning(s).

I report this in the last 5 years several times.
So, I hope you can fix this soon.
With regards,
Reto

ALABSTM, 2020-10-14:
Hi @CanastraRF,

The issue you reported has been confirmed. Actually, it has been already fixed internally. The fix will be available in a future release, very soon hopefully. Below is how the macro has been redefined.
```diff
#define __HAL_RTC_TAMPER_GET_IT(__HANDLE__, __INTERRUPT__)    (((__INTERRUPT__) == RTC_IT_TAMP1) ? (((((__HANDLE__)->Instance->ISR) & ((__INTERRUPT__)>> 3U)) != 0U) ? 1U : 0U) : \
                                                               ((__INTERRUPT__) == RTC_IT_TAMP2) ? (((((__HANDLE__)->Instance->ISR) & ((__INTERRUPT__)>> 5U)) != 0U) ? 1U : 0U) : \
-                                                               ((__INTERRUPT__) == RTC_IT_TAMP3) ? (((((__HANDLE__)->Instance->ISR) & ((__INTERRUPT__)>> 7U)) != 0U) ? 1U : 0U))
+                                                                                                   (((((__HANDLE__)->Instance->ISR) & ((__INTERRUPT__)>> 7U)) != 0U) ? 1U : 0U))
```
Thank you for your patience and thank you again for your report.

With regards,

CanastraRF, 2020-10-14:
Hi @ALABSTM 
Thank You for your feedback.
I Hope this will be fixed soon too.

I've got some improvments in the cube code.
Is this a way to communicate with the development?

The 1st level support often does not understand the problems.
Reto Felix

ALABSTM, 2020-10-14:
Hi @CanastraRF,

Indeed, this support channel has been especially created for all aspects related to the Cube firmware published within these repositories (CMSIS Device, HAL-LL, MW, BSP, and Projects). Your suggestions and proposals are more than welcome.

Please note however that any other topic related to documentation, ecosystem, or hardware for instance is out of the scope of this support channel. In that case, you may rather submit your questions and proposals to the [ST Community](https://community.st.com/s/).

I hope I answered your question. We are looking forward to reading your improvement proposals.

With regards,

ALABSTM, 2020-10-14:
ST Internal Reference: 58409

ALABSTM, 2021-03-10:
Hi @CanastraRF,

I hope you are doing well. Issue fixed and fix now available in the frame of the STM32CubeL0 **v1.12.0** published a couple of days ago on GitHub. Thank you again for having reported.

With regards,
Ali

closed 2021-03-10 by ALABSTM