STMicroelectronics/STM32CubeL0 #12: Add HAL_TIM_DMABurst_MultiWriteStart and HAL_TIM_DMABurst_MultiReadStart

statropy, 2020-10-27:
Other repositories have been updated with `HAL_TIM_DMABurst_MultiWriteStart` and `HAL_TIM_DMABurst_MultiReadStart`. Is there any reason these functions can't be added for STM32CubeL0 also?

ASELSTM, 2020-11-02:
Hi @statropy,

Actually, these functions has been added internally but not yet released. The fix will be published in the frame of a future publication. We cannot share a date for the moment. Stay tuned and thank you once more for your contribution and for your patience.

With regards,

ASELSTM, 2020-12-09:
Hi @statropy, 

I hope you are fine. The issue you reported has been fixed in the frame of version **v1.12.0** of the STM32CubeL0 published recently on GitHub.

Thank you again for having reported.

With regards,

closed 2020-12-09 by ASELSTM