STMicroelectronics/STM32CubeL0 #14: Build Error in STM32CubeL0_v1.12.0\stm32l0xx_hal_dac.c for STM32L052 and others

CanastraRF, 2021-03-20:
I download the  STM32CubeL0 V1.12.0
When I build stm32l0xx_hal_dac.c for STM32L052 with USE_HAL_DAC_REGISTER_CALLBACKS = 1 I get

stm32l0xx_hal_dac.c(296): error:  #20: identifier "HAL_DACEx_ConvCpltCallbackCh2" is undefined
      hdac->ConvCpltCallbackCh2           = HAL_DACEx_ConvCpltCallbackCh2;
stm32l0xx_hal_dac.c(297): error:  #20: identifier "HAL_DACEx_ConvHalfCpltCallbackCh2" is undefined
      hdac->ConvHalfCpltCallbackCh2       = HAL_DACEx_ConvHalfCpltCallbackCh2;
stm32l0xx_hal_dac.c(298): error:  #20: identifier "HAL_DACEx_ErrorCallbackCh2" is undefined
      hdac->ErrorCallbackCh2              = HAL_DACEx_ErrorCallbackCh2;
stm32l0xx_hal_dac.c(299): error:  #20: identifier "HAL_DACEx_DMAUnderrunCallbackCh2" is undefined
      hdac->DMAUnderrunCallbackCh2        = HAL_DACEx_DMAUnderrunCallbackCh2;
stm32l0xx_hal_dac.c(937): error:  #20: identifier "HAL_DACEx_ConvCpltCallbackCh2" is undefined
        hdac->ConvCpltCallbackCh2 = HAL_DACEx_ConvCpltCallbackCh2;
stm32l0xx_hal_dac.c(940): error:  #20: identifier "HAL_DACEx_ConvHalfCpltCallbackCh2" is undefined
        hdac->ConvHalfCpltCallbackCh2 = HAL_DACEx_ConvHalfCpltCallbackCh2;
stm32l0xx_hal_dac.c(943): error:  #20: identifier "HAL_DACEx_ErrorCallbackCh2" is undefined
        hdac->ErrorCallbackCh2 = HAL_DACEx_ErrorCallbackCh2;
stm32l0xx_hal_dac.c(946): error:  #20: identifier "HAL_DACEx_DMAUnderrunCallbackCh2" is undefined
        hdac->DMAUnderrunCallbackCh2 = HAL_DACEx_DMAUnderrunCallbackCh2;
stm32l0xx_hal_dac.c: 0 warnings, 8 errors

Please fix this ASAP with 
#if defined (STM32L072xx) || defined (STM32L073xx) || defined (STM32L082xx) || defined (STM32L083xx)
#endif
like in stm32l0xx_hal_dac_ex.h
ReFe

ASELSTM, 2021-03-22:
Hi @CanastraRF, 

Thank you for this other report.  A fix will be implemented and made available in the future release. Thank you once again for your contribution.

With regards,

ASELSTM, 2021-03-22:
ST Internal Reference: 103592

ASELSTM, 2021-08-16:
Hi @CanastraRF,

The issue you reported has been fixed in the frame of version v1.12.1 of the STM32CubeL0 published on GitHub.

Please allow me then to close this thread and thank you again for having reported.

With regards,

closed 2021-08-16 by ASELSTM