STMicroelectronics/STM32CubeL0 #15: RCC->CRRCR register accesses should have RCC_HSI48_SUPPORT compilation guard

ricehornet, 2021-04-02:
In the STM32L0 example projects' `system_stm32l0xx` modules there is a reconfiguration of the RCC->CRRCR register in the default clock config, but only MCUs with an HSI48 include this register. For example, I was using this project with a STM32L071CB MCU which does not include the RCC->CRRCR register:

https://github.com/STMicroelectronics/STM32CubeL0/blob/master/Projects/NUCLEO-L073RZ/Examples/PWR/PWR_STOP/Src/system_stm32l0xx.c#L146-L147

I think it should be:

```
#if defined(RCC_HSI48_SUPPORT)
/*!< Reset HSI48ON  bit */
  RCC->CRRCR &= (uint32_t)0xFFFFFFFEU;
#endif
```

While this didn't appear to cause any problems for me, I was confused when using the example because I couldn't find this register in my MCU's reference manual. Hoping that it can help others who are going through the code line-by-line.

RKOUSTM, 2021-04-16:
Hi @ricehornet,

Thank you for your contribution. Actually, the reported point and the other point raised in #16 have already been resolved. In fact, we have discussed them internally and we are working on aligning all STM32CubeL0 projects with the [system_stm32l0xx.c](https://github.com/STMicroelectronics/STM32CubeL0/blob/a7b74aed35ecb7baeadeb16107aa8fddb6823589/Drivers/CMSIS/Device/ST/STM32L0xx/Source/Templates/system_stm32l0xx.c#L154-L160) file. A fix will be made available in the frame of a future release.

Please allow me then to close this one and to consider the other. Stay tuned and thank you once more for your contribution.

With regards,

RKOUSTM, 2021-04-16:
ST Internal Reference: 104564

closed 2021-04-16 by RKOUSTM

RKOUSTM, 2021-04-19:
Duplicate of #16.