STMicroelectronics/STM32CubeL0 #16: MCO register is incorrect width in some SystemInit examples

ricehornet, 2021-04-02:
When using this example on an STM32L071CB:

https://github.com/STMicroelectronics/STM32CubeL0/blob/master/Projects/NUCLEO-L073RZ/Examples/PWR/PWR_STOP/Src/system_stm32l0xx.c#L140-L141

I noticed that MCOSEL bits [2:0] are reset, but this register is actually 4 bits wide on STM32L0 and in order to disable MCO (I assume that is the intention), we need to reset all the bits.

 Therefore I changed it to

```
RCC->CFGR &= (uint32_t) 0x80FF400CU;
```

[image]

RKOUSTM, 2021-04-08:
Hi @ricehornet,

Thank you for your contribution. The issue you pointed out has been confirmed. An internal tracker has been logged and a fix will be implemented and made available in the frame of a future release.

Thank you once again for your contribution.

With regards,

RKOUSTM, 2021-04-08:
ST Internal Reference: 104564

RKOUSTM, 2021-08-16:
Hi @ricehornet,

The fix you requested has been implemented and is now available in the frame of the latest STM32CubeL0 package V1.12.1 release. 

This issue can be closed now. Thank you again for your contribution.

With regards,

closed 2021-08-16 by RKOUSTM