STMicroelectronics/STM32CubeL0 #18: HAL_PWR_EnterSLEEPMode() HAL_PWR_EnterSTOPMode() and WFI instructions

apiesse, 2021-04-22:
Was pyaling around with low poer modes on a STM32L071KZ and I did found out that both HAL_PWR_EnterSLEEPMode() and HAL_PWR_EnterSTOPMode() use what looks like to be a wrong WFI sequence.

The functions have the following code (which is not working, uC not going into Sleep/Stop mode)
`  if(STOPEntry == PWR_STOPENTRY_WFI)
  {
    /* Request Wait For Interrupt */
    __WFI();
  }`

Changing to this code solved my issue:
`  if(STOPEntry == PWR_STOPENTRY_WFI)
  {
    /* Request Wait For Interrupt */
    __DSB();
    __WFI();
    __ISB();
  }`

ALABSTM, 2021-04-26:
Hi @apiesse,

Thank you for your report. From what I can notice, `__DSB();` and `__ISB();` instructions are only used in the drivers of the STM32F7 and STM32H7 series. As for other series, they are not. Do you confirm this issue you are facing with your STM32L071KZ µC is systematically reproducible? Thank you.

With regards,

apiesse, 2021-04-26:
Hi @ALABSTM,
I did checked the code again and found other issues which actually caused the initial problem entering STOP mode.
As you mentioned, `__DSB()` and `__ISB()` and not needed. I was misleaded by the FreeRTOS tickless implementation and some comments found on the web.

For what I can tell, the issue can be closed.

ALABSTM, 2021-04-28:
Hi @apiesse,

Glad to hear you could overcome the problem. I will close the issue.

With regards,

closed 2021-04-28 by ALABSTM