STMicroelectronics/STM32CubeL0 #19: Duplicate volatile reads in interrupt handler for hal_timer

jrahlf, 2021-04-26:
Similar to #17 , there are unnecessary duplicate volatile reads in the interrupt handler for the timer peripheral.
`__HAL_TIM_GET_FLAG` is called several times inside the handler, each time the actual value is fetched due to volatile qualifier.

https://github.com/STMicroelectronics/STM32CubeL0/blob/a7b74aed35ecb7baeadeb16107aa8fddb6823589/Drivers/STM32L0xx_HAL_Driver/Src/stm32l0xx_hal_tim.c#L3560

As in #17, this likely affects all other device variants (F1/F3/F4...) as well.

ASELSTM, 2021-06-08:
ST Internal Reference: 108610

ASELSTM, 2021-06-09:
Hi @jrahlf,

Thank you for your contribution. This has been communicated to the development team. We will come back to you as soon as we get their feedback.

Thank you again for your contribution.

With regards,

TOUNSTM, 2024-01-25:
Fixed in [9d7402e](https://github.com/STMicroelectronics/stm32l0xx_hal_driver/commit/9d7402ea840357d8224592af199a132c3340ee1f)

closed 2024-01-25 by TOUNSTM