STMicroelectronics/STM32CubeL0 #20: Duplicate volatile reads in interrupt handler for adc

jrahlf, 2021-04-26:
Similar to #17 , there are unnecessary duplicate volatile reads in the interrupt handler for the adc peripheral.
`__HAL_ADC_GET_FLAG`  is called several times inside the handler, each time the actual value is fetched due to volatile qualifier.

https://github.com/STMicroelectronics/STM32CubeL0/blob/a7b74aed35ecb7baeadeb16107aa8fddb6823589/Drivers/STM32L0xx_HAL_Driver/Src/stm32l0xx_hal_adc.c#L1692

As in #17, this likely affects all other device variants (F1/F3/F4...) as well. Interestingly, the implementation for the F4 variant actually caches the status register in a local variable, but does so multiple times, which defeats its purpose.

ASELSTM, 2021-06-09:
Hi @jrahlf,

Thank you for your contribution. This has been communicated to the development team. We will come back to you as soon as we get their feedback.

Thank you again for your contribution.

With regards,

ALABSTM, 2021-11-09:
ST Internal Reference: 111909

RJMSTM, 2023-03-07:
Hello @jrahlf 

I hope you are fine.

The issue you reported has been fixed in the frame of version v1.12.2 of the STM32CubeL0 published recently on GitHub.

Thank you again for having reported.

BeST Regards,
Rania