STMicroelectronics/STM32CubeL0 #21: HAL_MPU_ConfigRegion() and HAL_MPU_Enable() barrier instructions are not compliant with the ARM specs

rbsexton, 2021-05-30:
HAL_MPU_ConfigRegion() is missing barrier instructions. (STM32CubeL0/Drivers/STM32L0xx_HAL_Driver/Src/stm32l0xx_hal_cortex.c)  

The companion function, void HAL_MPU_Enable(uint32_t MPU_Control) Includes barrier instructions. 

Per Arm Cortex-M0+ DevicesGeneric User Guide ( DUI0662B_cortex_m0p_r0p1_dgug.pdf), Section 4.5.8:

> Software must use memory barrier instructions:

> - Before MPU setup if there might be outstanding memory transfers, such as buffered writes, that might be affected by the change in MPU settings.
> - After MPU setup, if the software includes memory transfers that must use the new MPU settings.

> However, an instruction synchronization barrier instruction is not required if the MPU setup process starts by entering an exception handler, or is followed by an exception return, because the exception entry and exception return mechanisms cause memory barrier behavior.

> For example, if you want all of the memory access behavior to take effect immediately after the programming sequence, use a DSB instruction and an ISB instruction. A DSB is required after changing MPU settings, such as at the end of context switch. An ISB is required if the code that programs the MPU region or regions is entered using a branch or call. If the programming sequence is entered using a return from exception, or by taking an exception, then you do not require an ISB.

If the Architecture of the L0 is such that that these barriers are not necessary or if the user is expected to supply them ( perhaps because they are being used within an ISR) that fact should be documented and/or clarified with expected usage samples.

RKOUSTM, 2021-07-01:
Hi @rbsexton,

Thank you for your reports. Your request has been forwarded to our development teams. We will be back to you as soon as we get an answer.

Thank you again for you reports.

With regards,

RKOUSTM, 2021-09-01:
Hi @rbsexton,

Thank you for your report. An internal tracker has been logged and a fix will be implemented and made available in the frame of a future release.

Thank you once again for your contribution.

With regards,

RKOUSTM, 2021-09-01:
ST Internal Reference: 112614

RJMSTM, 2023-03-07:
Hello @rbsexton 

I hope you are fine.

The issue you reported has been fixed in the frame of version v1.12.2 of the STM32CubeL0 published recently on GitHub.

Thank you again for having reported.

BeST Regards,
Rania

closed 2023-03-07 by RJMSTM