STMicroelectronics/STM32CubeL0 #22: Unused parameter in Drivers/STM32L0xx_HAL_Driver/Inc/stm32l0xx_ll_adc.h

RobinMaub, 2021-07-06:
Hello ST team,

The function `LL_ADC_DMA_GetRegAddr` contained in _Drivers/STM32L0xx_HAL_Driver/Inc/stm32l0xx_ll_adc.h_ does not use the `Register` parameter, leading to warnings.

Please implement `UNUSED(Register);` to clear them :)

Best regards

ASELSTM, 2021-07-12:
Hi @RobinMaub,

The issue you pointed out is confirmed. A fix will be implemented and made available in future release. Thank you once again for your contribution.

With regards,

ASELSTM, 2021-07-12:
ST Internal Reference: 110507

ASELSTM, 2021-08-16:
Hi @RobinMaub,

The issue you reported has been fixed in the frame of version v1.12.1 of the STM32CubeL0 published on GitHub.

Please allow me then to close this thread and thank you again for having reported.

With regards,

closed 2021-08-16 by ASELSTM