STMicroelectronics/STM32CubeL0 #25: USBD_CUSTOM_HID_SendReport() not working

factoryal, 2022-05-05:
Hi.

I'm using STM32L053R8T6 based custom PCB development board and testing USB custom HID.
the code
```c
errno = USBD_CUSTOM_HID_SendReport(&hUsbDeviceFS, usb_send_data, sizeof(usb_send_data));
```
does not works and returned value `errno` is `1` which represents `USBD_BUSY`.
I used USB HID Demonstrator v1.0.2 tool to monitor it's communication status.

I solved this problem with simply downgrade to firmware package version 1.12.0 and it works as expected.
`USBD_CUSTOM_HID_SendReport()` function returns `0`(`USBD_OK`)

Please check this issue.
Thanks.

factoryal, 2022-05-15:
I discovered USB HID keyboard not working properly.
Also can be fixed by downgrade version to 1.12.0

Changing the title to "USB Sending Report not working".

Tested code:
```cpp
uint8_t test[8] = { 0, };
while(1) {
  test[2] = 0x11;
  USBD_HID_SendReport(&hUsbDeviceFS, (uint8_t *)&test, sizeof(test));
  HAL_Delay(30);
  test[2] = 0x00;
  USBD_HID_SendReport(&hUsbDeviceFS, (uint8_t *)&test, sizeof(test));
  HAL_Delay(1000);
}
```

ch-f, 2022-08-19:
Doing tutorial [09.7_USB HID device - custom device lab - solution.zip](https://www.st.com/content/st_com/en/support/learning/stm32-education/stm32-moocs/STM32-USB-training.html) ([reference video](https://www.youtube.com/watch?v=3JGRt3BFYrM)) on firmware 1.12.1 fails, but works on 1.12.0:  HID_terminal.exe works two times (sends and recieves data by clicking on Send-Button) but fails reproducible after second time with an error "Operation timed out".

ch-f, 2022-09-04:
```
--- a/Drivers/STM32L0xx_HAL_Driver/Src/stm32l0xx_hal_pcd.c
+++ b/Drivers/STM32L0xx_HAL_Driver/Src/stm32l0xx_hal_pcd.c
@@ -1851,28 +1851,6 @@ static HAL_StatusTypeDef PCD_EP_ISR_Handler(PCD_HandleTypeDef *hpcd)
         /* clear int flag */
         PCD_CLEAR_TX_EP_CTR(hpcd->Instance, epindex);
 
-        if (ep->type != EP_TYPE_BULK)
-        {
-          ep->xfer_len = 0U;
-
-          if ((wEPVal & USB_EP_DTOG_TX) != 0U)
-          {
-            PCD_SET_EP_DBUF0_CNT(hpcd->Instance, ep->num, ep->is_in, 0U);
-          }
-          else
-          {
-            PCD_SET_EP_DBUF1_CNT(hpcd->Instance, ep->num, ep->is_in, 0U);
-          }
-
-          /* TX COMPLETE */
-#if (USE_HAL_PCD_REGISTER_CALLBACKS == 1U)
-          hpcd->DataInStageCallback(hpcd, ep->num);
-#else
-          HAL_PCD_DataInStageCallback(hpcd, ep->num);
-#endif /* USE_HAL_PCD_REGISTER_CALLBACKS */
-        }
-        else
-        /* Manage Bulk Single Buffer Transaction */
         if ((ep->type != EP_TYPE_BULK) ||
                              ((ep->type == EP_TYPE_BULK) && ((wEPVal & USB_EP_KIND) == 0U)))
         {

```
My issue: Only first two HID URBs are getting through, after applying this patch on firmware 1.12.1 it seems to make it work just fine. But I have no clue why. Could you please check on this?

factoryal, 2022-09-05:
@ch-f Thanks for your comment.
Unfortunately, your patch didn't resolved my problem.
It still cannot send any packet to host.

ch-f, 2022-09-07:
The code there seems obviously wrong, it does:
```
    if (a) {
    ...
    } else if (a || b) {
     ...
    }
```
There might be other traps like this or you might need the bottom part after the 'else if' of (ep->type != EP_TYPE_BULK)...

Would be nice if an official from ST could shed light?