STMicroelectronics/STM32CubeL0 #27: Adding user context to the peripheral handlers

hippiefm, 2022-11-06:
Hello,

I am writing a C++ wrapper on the top of ST's HAL and I have encountered a problem connected with interrupt callbacks and member functions. 

**Introduction**

Registering interrupt callbacks is really cool thing in HAL, but it is not possible to assign a class member function as a callback function (it needs to be static). This problem is solved inside FreeRTOS, as it is possible to pass a user context to a task and then wrap handler function like this:

```
static void StaticThreadWrapper(void *parameter) {
            auto t = static_cast<ThreadController *>(parameter);
            t->ThreadTask();
}
```

Same thing is not possible inside HAL, but adding simple user context inside peripheral handler would fix the issue and make C++ developer's life easier.

**Proposed solution**

Add user context inside peripheral handlers (UART as example, but all other peripherals should be handled the same way).

```
void                          *pUserPtr;        /*!< Pointer to user context            */
```
After change:

```
/**
  * @brief  UART handle Structure definition
  */
typedef struct __UART_HandleTypeDef
{
  USART_TypeDef                 *Instance;        /*!< UART registers base address        */

  UART_InitTypeDef              Init;             /*!< UART communication parameters      */

  uint8_t                       *pTxBuffPtr;      /*!< Pointer to UART Tx transfer Buffer */

  uint16_t                      TxXferSize;       /*!< UART Tx Transfer size              */

  __IO uint16_t                 TxXferCount;      /*!< UART Tx Transfer Counter           */

  uint8_t                       *pRxBuffPtr;      /*!< Pointer to UART Rx transfer Buffer */

  uint16_t                      RxXferSize;       /*!< UART Rx Transfer size              */

  __IO uint16_t                 RxXferCount;      /*!< UART Rx Transfer Counter           */

  __IO HAL_UART_RxTypeTypeDef ReceptionType;      /*!< Type of ongoing reception          */

  DMA_HandleTypeDef             *hdmatx;          /*!< UART Tx DMA Handle parameters      */

  DMA_HandleTypeDef             *hdmarx;          /*!< UART Rx DMA Handle parameters      */

  HAL_LockTypeDef               Lock;             /*!< Locking object                     */

  __IO HAL_UART_StateTypeDef    gState;           /*!< UART state information related to global Handle management
                                                       and also related to Tx operations.
                                                       This parameter can be a value of @ref HAL_UART_StateTypeDef */

  __IO HAL_UART_StateTypeDef    RxState;          /*!< UART state information related to Rx operations.
                                                       This parameter can be a value of @ref HAL_UART_StateTypeDef */

  __IO uint32_t                 ErrorCode;        /*!< UART Error code                    */

  void                          *pUserPtr;        /*!< Pointer to user context            */ 

#if (USE_HAL_UART_REGISTER_CALLBACKS == 1)
  void (* TxHalfCpltCallback)(struct __UART_HandleTypeDef *huart);        /*!< UART Tx Half Complete Callback        */
  void (* TxCpltCallback)(struct __UART_HandleTypeDef *huart);            /*!< UART Tx Complete Callback             */
  void (* RxHalfCpltCallback)(struct __UART_HandleTypeDef *huart);        /*!< UART Rx Half Complete Callback        */
  void (* RxCpltCallback)(struct __UART_HandleTypeDef *huart);            /*!< UART Rx Complete Callback             */
  void (* ErrorCallback)(struct __UART_HandleTypeDef *huart);             /*!< UART Error Callback                   */
  void (* AbortCpltCallback)(struct __UART_HandleTypeDef *huart);         /*!< UART Abort Complete Callback          */
  void (* AbortTransmitCpltCallback)(struct __UART_HandleTypeDef *huart); /*!< UART Abort Transmit Complete Callback */
  void (* AbortReceiveCpltCallback)(struct __UART_HandleTypeDef *huart);  /*!< UART Abort Receive Complete Callback  */
  void (* WakeupCallback)(struct __UART_HandleTypeDef *huart);            /*!< UART Wakeup Callback                  */
  void (* RxEventCallback)(struct __UART_HandleTypeDef *huart, uint16_t Pos); /*!< UART Reception Event Callback     */

  void (* MspInitCallback)(struct __UART_HandleTypeDef *huart);           /*!< UART Msp Init callback                */
  void (* MspDeInitCallback)(struct __UART_HandleTypeDef *huart);         /*!< UART Msp DeInit callback              */
#endif  /* USE_HAL_UART_REGISTER_CALLBACKS */

} UART_HandleTypeDef;
```

ASELSTM, 2023-02-17:
Hi @hippiefm,

Hi @jrahlf,

Thank you for your contribution. Your request has been communicated to the development team. We will come back to you as soon as we get their feedback.

Thank you again for your contribution.

With regards,

ASELSTM, 2023-02-17:
ST Internal Reference: 145785

ASELSTM, 2023-04-10:
Hi @hippiefm,

Unfortunately, there is no plan to deploy this in the near future. We will keep you informed, should there be any update in this plan.

Please allow me to close this issue. Thank you for your comprehension and thank you again for your report.

With regards,

closed 2023-04-10 by ASELSTM