STMicroelectronics/STM32CubeL0 #29: Presence of TS_CAL1 on STM32L011

badcf00d, 2022-11-25:
The LL API seems to incorrectly assume that TS_CAL1 (temperature sensor calibration value @ 30c) is missing on the STM32L011 series.

The datasheet says in section 6.3.16 Temperature Sensor Characteristics, sub-point 2, "V30 ADC conversion result is stored in the TS_CAL1 byte".

I've tested this on STM32L011 development boards, and the TS_CAL1 value is indeed present in 0x1FF8007A, and the value seems sensible, producing quite accurate temperature readings using the included macro.

Could stm32l0xx_ll_adc.h be modified to correct this? Thanks.

RJMSTM, 2022-12-26:
Hello @badcf00d 

Would you please give us more details about how you got this issue? 
Could you please share a screenshot of the value read in addresses 0x1FF8007A (TS_CAL1)  or share the project you have used to reproduce the issue in order to allow a better analysis of the problem.

With regards,
Rania

badcf00d, 2023-01-06:
Hi @RJMSTM

The issue is just that there are multiple comments and #defines in `stm32l0xx_ll_adc.h` that say that TS_CAL1 isn't available on STM32L011:

https://github.com/STMicroelectronics/STM32CubeL0/blob/bec4e499a74de98ab60784bf2ef1912bee9c1a22/Drivers/STM32L0xx_HAL_Driver/Inc/stm32l0xx_ll_adc.h#L196-L198 and
https://github.com/STMicroelectronics/STM32CubeL0/blob/bec4e499a74de98ab60784bf2ef1912bee9c1a22/Drivers/STM32L0xx_HAL_Driver/Inc/stm32l0xx_ll_adc.h#L1461-L1467

But this seems to be incorrect. `*((uint16_t*) 0x1FF8007AU)` on one of my dev boards is `649`, another it's `664`, and they both give accurate temperature readings when I remove the #defines and use the `__LL_ADC_CALC_TEMPERATURE` macro.

So I think the fix is just to remove the comments about TS_CAL1 being unavailable on L011, and remove `STM32L011xx` from the #defines that I mentioned above.

Thanks,

badcf00d, 2023-05-19:
Hi @RJMSTM,

Thanks for getting back to me, like I say it seems like the TS_CAL1 value is present on the L011, all the dev boards I've got give accurate temperature within a couple of degrees when using it. So yeh maybe it's just been left off the datasheet by accident or something.

Regards,

RJMSTM, 2023-08-08:
Hello @badcf00d 

Please allow me to close this thread as no activity. You may reopen it at any time if you have any details to share with us in order to help you to solve the issue. Thank you for your comprehension.

With regards

badcf00d, 2023-08-08:
I don't think this is ready to be closed? This still seems to be a bug in the LL API:

https://github.com/STMicroelectronics/STM32CubeL0/blob/e416b1d1ede2719bc253eb8c8029e8e338567f2f/Drivers/STM32L0xx_HAL_Driver/Inc/stm32l0xx_ll_adc.h#L195C81-L195C81

If you would like any more information let me know, but basically, `0x1FF8007AU` seems to hold TS_CAL1 on L011, but the LL API is wrong and says TS_CAL1 is not present on L011.