STMicroelectronics/STM32CubeL0 #33: IS_LPTIM_PERIOD() lower value should be 0

DaniAtGH, 2023-11-20:
The Period start value has changed from 0 to 1 in the macro IS_LPTIM_PERIOD(). In my opinion the lowest autoreload value should be 0.
I have tested a 1s period with the external LSE clock 32,768kHz and STM32L071CB: Timeout = 0x7FFF, due to processor errata = 0x7FFE, HAL_LPTIM_TimeOut_Start_IT(&lptimHandle, 1U, 0x7FFEU)). The LPTIM interrupt is raised every 999,95ms and before (with the HAL version 1.10.5) every 999,99ms (HAL_LPTIM_TimeOut_Start_IT(&lptimHandle, 0U, 0x7FFEU)).
Same issue is also in STM32CubeL4.

MKISTM, 2025-11-27:
Hello @DaniAtGH,

Thank you for your intervention, I'll proceed with closing this issue, thank you for your understanding.

Please let us know if you encounter any specific unexpected behavior or errors, and we will be happy to assist further.

BR,
Mohamed Chaker.

closed 2025-12-04 by MKISTM

DaniAtGH, 2025-12-11:
Have you considered the device errata ES0292, errata 2.10.3?
Over 2 years ago, when I opened this issue:
I have tested a 1s period with the external LSE clock 32,768kHz and STM32L071CB: Timeout = 0x7FFF, due to processor errata = 0x7FFE, HAL_LPTIM_TimeOut_Start_IT(&lptimHandle, 1U, 0x7FFEU)). The LPTIM interrupt is raised every 999,95ms and before (with the HAL version 1.10.5) every 999,99ms (HAL_LPTIM_TimeOut_Start_IT(&lptimHandle, 0U, 0x7FFEU)).
Same issue is also in STM32CubeL4.