STMicroelectronics/STM32CubeL0 #34: usbd_cdc_if.c lacking function definition for CDC_TransmitCplt_FS

kaptptrx, 2024-02-14:
The file `USB_DEVICE/App/usbd_cdc_if.c` in the stm32L0 HAL, as generated by CubeMX doesn't contain support registering a callback for USB transmit complete, whereas this functionality does exist in stm32L4, among other HALs. Instead of registering a callback, we're forced to call a function in the application main loop.

This is the function that is defined in L4 but not L0.
```
static int8_t CDC_TransmitCplt_FS(uint8_t *pbuf, uint32_t *Len, uint8_t epnum);
````

It looks like the lower level USB HAL supports this callback, and the L0 reference manual indicates the transmit complete interrupt is supported. Curious as to why it is not supported in the USB_DEVICE layer. 

Our team has considered adding this functionality to the `usbd_cdc_if.c`  file, but would prefer not to maintain code instead of just using autogenerated CubeMX code. 

Is this a feature that you could add to the stm32CubeL0 HAL? Or if our team created a pull request adding this feature would you consider adding?

Thanks.

ALABSTM, 2024-02-19:
Hi @kaptptrx,

I had a look into [STM32CubeL4](https://github.com/STMicroelectronics/STM32CubeL4) firmware and could not find the `CDC_TransmitCplt_FS()` callback you mentioned. Rather, I found mention of a `CDC_Itf_TransmitCplt()` callback [here](https://github.com/STMicroelectronics/STM32CubeL4/blob/2a8a13bc3c2623147d982aa1dbd3a13afd88ebc1/Projects/NUCLEO-L452RE/Applications/USB_Device/CDC_Standalone/Src/usbd_cdc_interface.c#L298-L305) for instance. I suppose they refer to the same thing.

Anyway, your question has been forwarded to our development teams. I will keep you updated.

With regards,

kaptptrx, 2024-02-20:
Hi @ALABSTM 

Thanks for your response. 

It looks like the `usbd_cdc_if.c` file comes from the `usbd_cdc_if_template.c` file? The L4 version of that file contains a definition for  `static int8_t TEMPLATE_TransmitCplt(uint8_t *pbuf, uint32_t *Len, uint8_t epnum);` whereas the L0 version does not.

ALABSTM, 2024-11-21:
Hi @kaptptrx,

I had a second look to the **STM32CubeL4** and confirm the existence of the callback you are referring to in the [usbd_cdc_if_template.c](https://github.com/STMicroelectronics/STM32CubeL4/blob/cd403e258854650aba75f694b431a50ef211d154/Middlewares/ST/STM32_USB_Device_Library/Class/CDC/Src/usbd_cdc_if_template.c#L70) file.

This is due to the fact that the **STM32CubeL4** integrates version **2.11.3** of the USB Device library (formerly version 2.11.1 by the time you posted this issue), while the **STM32CubeL0** still integrates version **2.5.3**.

Unfortunately, there is no short-term plan for the **STM32CubeL0** firmware to support this newer version. Maybe in the long run. I will keep you informed, should there be any change.

As a workaround, may I suggest you manually port the needed files to your application or just add the needed callback? You can find the library sources in [this](https://github.com/STMicroelectronics/stm32-mw-usb-device) repository.

We apologize for any inconvenience. Please allow me to close this issue. Thank you for your comprehension.

With regards,

closed 2024-11-21 by ALABSTM