STMicroelectronics/STM32CubeL0 #35: FatFs is not compatible with CMSIS_OS2 (osSemaphoreCreate is changed to osSemaphoreNew)

MaxFlatline, 2024-09-30:
**Set-Up**
Stm32l072czt
STM32CubeL0 v1.12.2

**Describe the bug**
syscall.c file contains old function names osSemaphoreCreate and  osSemaphoreWait which are renamed in Cmsis_os2

**How To Reproduce**
1. Choose Cmsis_os2 Freertos
2. Select User-defined FatFS
3. Error in build

**Screenshots**
[image]

ASEHSTM, 2024-10-04:
Hello @MaxFlatline,

Currently, the version of FatFS on the STM32L0 does not support CMSIS-RTOS v2. I have shared this proposal with the team and will keep you informed if there are any updates.

Thank you for your understanding.
With Regards,

ASEHSTM, 2024-10-29:
Hello @MaxFlatline,

Currently, there are no plans to update the version of FatFS on the STM32L0. Therefore, please allow me to close this issue.

Thank you for your understanding.
With Regards,

closed 2024-10-29 by ASEHSTM

MaxFlatline, 2024-10-29:
Oh my god, guys, it's like three lines of conditionals to make them compatible. Then make some message about compatibility in cubeMX, I don't know. Thanks, but no thanks, I guess.